bigtreetech/PandaTouch #309: v1.0.7.2 issues

JohnnyofRC, 2025-05-30:
Fixed BUGS
add v1.0.7.2

    1 fixed bug
        crash when download thumbnails from history
    **2 new feature
        custom filament
        reprint with custom filament**
    3 modify backlight
        minimum is 1% and maximum is 80%.
        set the backlight to 1% when power-on and set the backlight to normal after 8 seconds.

Point 2 is NOT working, reprint feature does nothing on the Panda Touch.

RobPFarley, 2025-05-31:
> > I'm having the same issue and I'm slicing with bambu studio, when selecting reprint the bed and nozzle heat up but then nothing happens after
> 
> Hi,
> Could you please send me your account ?
> You can change the password to a simple one and I will do a test with it as soon as possible and I will try my best to solve this problem.
> and you can chang password back after testing.
> My email is 1284513393@qq.com.
> 
> I will do this next Tuesday.
> 

Sorry, I'm not giving my account details to a stranger on the internet

dilixi, 2025-05-31:
> > > I'm having the same issue and I'm slicing with bambu studio, when selecting reprint the bed and nozzle heat up but then nothing happens after
> > 
> > 
> > Hi,
> > Could you please send me your account ?
> > You can change the password to a simple one and I will do a test with it as soon as possible and I will try my best to solve this problem.
> > and you can chang password back after testing.
> > My email is [1284513393@qq.com](mailto:1284513393@qq.com).
> > I will do this next Tuesday.
> 
> Sorry, I'm not giving my account details to a stranger on the internet

It is okay.
I don't have a oversea printer so I need someone to help me to solve this problem.
But I am not sure we can reproduce it from other kind people.

RobPF92, 2025-05-31:
> > > > I'm having the same issue and I'm slicing with bambu studio, when selecting reprint the bed and nozzle heat up but then nothing happens after
> > > 
> > > 
> > > Hi,
> > > Could you please send me your account ?
> > > You can change the password to a simple one and I will do a test with it as soon as possible and I will try my best to solve this problem.
> > > and you can chang password back after testing.
> > > My email is [1284513393@qq.com](mailto:1284513393@qq.com).
> > > I will do this next Tuesday.
> > 
> > Sorry, I'm not giving my account details to a stranger on the internet
> 
> I like to solve problem but I do not have a better way but ask an account.

Do you work for bigtreetech? There has to be a better way to debug than to ask for users' personal account details

dilixi, 2025-05-31:
> Do you work for bigtreetech? There has to be a better way to debug than to ask for users' personal account details

Do you work for bigtreetech?
Yes.

You are right，Thank you.

RobPF92, 2025-05-31:
> Do you work for bigtreetech?
> Yes.
>
> You are right，Thank you.

I hope I don't come across as rude when I say it's quite concerning that an employee of the company is asking for the login details of bambu lab user accounts. It's very unprofessional

RobPF92, 2025-06-01:
> > > > I hope I don't come across as rude when I say it's quite concerning that an employee of the company is asking for the login details of bambu lab user accounts. It's very unprofessional
> > > 
> > > 
> > > He's our primary firmware engineer on the Panda Touch. Had he asked for your final login details that you intend to continue using, we would be in full agreement. However, if you read his message you'll notice that he asked for the password to be temporarily changed to something simple so that he can see why your machine is not responding to the reprint command, all the while you would be welcome to monitor along as he works. Due to the nature of the MQTT communication this is the only real way to conduct debugging with users who are having problems that cannot be reproduced locally. As soon as the debugging is complete, he would immediately let you know and you would change the password to whatever you want it to be. Bear in mind that every login requires a 2fa code sent via email so even with the password he cannot do anything without you also issuing a code which means that you still get to control his access while he is debugging.
> > > He has used this method with others willing to help in many instances and been able to solve many bugs.
> > 
> > Regardless of the reason behind him asking for my login details, the fact remains that I would be putting my bambu lab account in the hands of a stranger, essentially giving them access to my print history, address details, order history and a lot of other personal data that I shouldn't be expected to share for any reason. The fact that I'm only temporarily granting access doesn't make this any less of a privacy risk, and as soon as I provide the 2fa code my account is no longer within my control. You shouldn't expect your users to give you login details for any third party accounts. If you truly have no other way of debugging then you should make it a priority to change your workflow to make it so such a security risk isn't needed.
> > 
> > In the interest of trying to get this bug fixed here's a photo of the error message that shows on my panda touch when I press reprint.
> > 
> > [image]
> > 
> > However after this exchange I'm now reluctant to continue using your product at all, as I'm getting the impression that your company doesn't take privacy seriously.
> 
> I just know I need to solve this problem for you and I am not interest to the privacy of anyone.
> 
> If you want to solve this problem you can email me anytime and I need to know something of your printer.

My printer is a P1S on firmware version 01.08.01.00. My panda touch is on firmware version 1.0.7.2. I sliced a file using bambu studio on the 0.2mm preset and sent it to the printer through the cloud. After the print finished I pressed reprint on my panda touch. The printer heated the bed and nozzle but didn't start the print, then after a few minutes the panda touch presented me with the above error.

I'm not giving you my bambu account login details, and you shouldn't expect anybody else to do so either. Your customers' privacy and security should be a top priority for you and asking for my bambu login is unacceptable. I'm a software developer myself and I would never dream of asking one of my customers to give me third party login details.

dilixi, 2025-06-01:
> > > > > I hope I don't come across as rude when I say it's quite concerning that an employee of the company is asking for the login details of bambu lab user accounts. It's very unprofessional
> > > > 
> > > > 
> > > > He's our primary firmware engineer on the Panda Touch. Had he asked for your final login details that you intend to continue using, we would be in full agreement. However, if you read his message you'll notice that he asked for the password to be temporarily changed to something simple so that he can see why your machine is not responding to the reprint command, all the while you would be welcome to monitor along as he works. Due to the nature of the MQTT communication this is the only real way to conduct debugging with users who are having problems that cannot be reproduced locally. As soon as the debugging is complete, he would immediately let you know and you would change the password to whatever you want it to be. Bear in mind that every login requires a 2fa code sent via email so even with the password he cannot do anything without you also issuing a code which means that you still get to control his access while he is debugging.
> > > > He has used this method with others willing to help in many instances and been able to solve many bugs.
> > > 
> > > 
> > > Regardless of the reason behind him asking for my login details, the fact remains that I would be putting my bambu lab account in the hands of a stranger, essentially giving them access to my print history, address details, order history and a lot of other personal data that I shouldn't be expected to share for any reason. The fact that I'm only temporarily granting access doesn't make this any less of a privacy risk, and as soon as I provide the 2fa code my account is no longer within my control. You shouldn't expect your users to give you login details for any third party accounts. If you truly have no other way of debugging then you should make it a priority to change your workflow to make it so such a security risk isn't needed.
> > > In the interest of trying to get this bug fixed here's a photo of the error message that shows on my panda touch when I press reprint.
> > > [image]
> > > However after this exchange I'm now reluctant to continue using your product at all, as I'm getting the impression that your company doesn't take privacy seriously.
> > 
> > 
> > I just know I need to solve this problem for you and I am not interest to the privacy of anyone.
> > If you want to solve this problem you can email me anytime and I need to know something of your printer.
> 
> My printer is a P1S on firmware version 01.08.01.00. My panda touch is on firmware version 1.0.7.2. I sliced a file using bambu studio v2.0.3.54 on the 0.2mm preset and sent it to the printer through the cloud. After the print finished I pressed reprint on my panda touch. The printer heated the bed and nozzle but didn't start the print, then after a few minutes the panda touch presented me with the above error.
> 
> I'm not giving you my bambu account login details, and you shouldn't expect anybody else to do so either. Your customers' privacy and security should be a top priority for you and asking for my bambu login is unacceptable. I'm a software developer myself and I would never dream of asking one of my customers to give me third party login details.

Can you tell me something about the filaments of your printer?

RobPF92, 2025-06-01:
> Can you tell me something about the filaments of your printer?

The filament I was trying to print with was a white sunlu pla+ that I have saved as generic white pla. The other filaments currently loaded are a black and blue sunlu pla+ and a bambu silk blue pla.

[image]

[image]

dilixi, 2025-06-01:
> The filament I was trying to print with was a white sunlu pla+ that I have saved as generic white pla. The other filaments currently loaded are a black and blue sunlu pla+ and a bambu silk blue [...]

Okay.
Thanks.
I will do the same thing with my PandaTouch.

JohnnyofRC, 2025-06-02:
> Be sure that you are slicing with bambu and not orca. I am not directly on this project but I think that may be a requirement for the reprint function.

We only use BAMBU STUDIO, no other slicer, we have 20 printers with this issue :(
Any fix for it ?

JohnnyofRC, 2025-06-06:
**Guys ? Anyone ?**