bigtreetech/PandaTouch #310: P1S Version 01.08.02.00 Update Information - New Firmware FYI

macman-3D, 2025-06-03:
# # New Features

# # # # 1. Authorization Control for Enhanced Security

- The printer now verifies whether control commands originate from official or
authorized software. If not, the commands will be rejected.
- You can still:
- Use third-party slicers with Bambu Connect.
- Operate the printer completely offline.
- Send files directly to the printer in LAN mode, without cloud connectivity
requirements.
- Downgrade firmware to a previous version if needed.
#### 2. Developer Mode (Available in "LAN Only" Mode).
- When enabled, Authorization Control will be disabled, allowing the printer to
accept all control instructions without verification.

- This mode is intended for advanced users who want full control over their
printer's security.

I'm staying v01.08.01.00 - until this update gets all figured out with Panda Touch.

Winky1st, 2025-06-03:
damn! 
Huston, we have a problem! :(

dzqc, 2025-06-04:
The fundamental problem can be solved by the cooperation between Bambu and Biqu. 👍

livegrenier, 2025-06-05:
Hey, I updated my PS1 to 01.08.02.00 and it's working fine, my Panda touch works just like it did before, I am on the latest Panda touch firmware, I'm not sure what the concerns are about in the comments? And in any case the printer has a firmware rollback option from the mobile app, if it stopped working you could go back...

[image]

[image]

bigtreetech, 2025-06-05:
> Hey, I updated my P1S to 01.08.02.00 and it's working fine, my Panda touch works just like it did before, I am on the latest Panda touch firmware, I'm not sure what the concerns are about in the comments? And in any case the printer has a firmware rollback option from the mobile app, if it stopped working you could go back...
> 
> [image]
> 
> [image]

Are you using LAN-mode with the developer option enabled?

Enclave10, 2025-06-05:
> Эй, я обновил свой P1S до версии 01.08.02.00, и он работает нормально, мой Panda touch работает так же, как и раньше, у меня последняя версия прошивки Panda touch, я не понимаю, о чём вы беспокоитесь в комментариях? В любом случае, в мобильном приложении есть возможность отката прошивки, если она перестала работать, вы можете вернуться...
> 
> [image]
> 
> [image]

у него облако отключено водит в заблуждение людей не работает значит авторизация Panda touch

Winky1st, 2025-06-05:
> > Are you using LAN-mode with the developer option enabled?
> 
> No, still using the default connection way with no changes

Have you tested read only functions or did you try to issue  some commands too from pandatouch? try to home your printer from panda touch ...shouldn't run.
 cheers.

coffeeandubuntu, 2025-06-05:
Winky1st is correct -- On the new firmware I was able to get read-only options on my Panda Touch but homing or pre-heating the bed using Panda Touch through errors. I've gone LAN-Only/Developer Mode and am trying out the Bambu Companion app as an alternative to Bambu Handy.

EddyK69, 2025-06-06:
I'm on on latest P1S firmware 01.08.02.00 and Panda Touch 1.0.7.3 and i can confirm that Panda Touch is still working (light, homing, heating etc). For this to work, you have to enable BOTH "Lan Only Mode" AND "Developer Mode".
See https://wiki.bambulab.com/en/knowledge-sharing/enable-developer-mode

Winky1st, 2025-06-06:
> I'm on on latest P1S firmware 01.08.02.00 and Panda Touch 1.0.7.3 and i can confirm that Panda Touch is still working (light, homing, heating etc). For this to work, you have to enable BOTH "Lan [...]

yes buddy that is clear from long time.. on lan mode it works ...but sadly on lan mode! only. i guess 90% of ppl dont use lan mode. ;) cheers

EddyK69, 2025-06-06:
> yes buddy that is clear from long time.. on lan mode it works ...but sadly on lan mode only! I guess 90% of ppl dont want to use lan mode. ;) cheers

Ah ok, I'm sorry for my unuseful comment then. 🫤

Winky1st, 2025-06-06:
> > yes buddy that is clear from long time.. on lan mode it works ...but sadly on lan mode only! I guess 90% of ppl dont want to use lan mode. ;) cheers
> 
> Ah ok, I'm sorry for my unuseful comment then. 🫤

Notthing is unuseful bro.  have fun.
cheers

Winky1st, 2025-06-06:
> all people downgrade or not upgrade. This is a road...

yeh ...until there aren't fw enhancements/upgrades. upto now the only difference with the new fw is just this lock; but what will happen once fw will have new features and improvements? 
I do really hope BigTreetech will  find a suitable solution with BBlab, differently I would really miss panda touch is too handy imo.