bigtreetech/PandaTouch #315: I2C Wireless Option for Panda Touch

GameWatchProjects, 2025-06-19:
Hello,

i use the Panda Touch in another room as the Bambulab Printer and I search for an i2C Wireless method to transfer the temperature and the humidify of the print chamber to the Panda Touch.

So i've found a solution to grab the I2C Signal and put them to WLAN, but found no solution to send them to Panda Touch.

If have anyone an idea for this problem?

CatCw, 2025-06-23:
Hi, you can put a module in the printer that acquires the temperature and the humidify from seneor, make a module disguised as AHT20-F, then send the data to PandaTouch.

GameWatchProjects, 2025-07-16:
@CatCw 

You have don't understand my problem. I can't use a **_wire_** option to connect a sensor to the Panda touch, while both equipment (Printer A1 and Panda touch) on different places and rooms in the house. So i looking for a method for a wireless I2C transfer, such like this as example: https://www.horter-shop.de/en/i2c-master-modules/256-477-kit-i2c-wlan-module-for-the-wemos-d1-4260404260936.html#/25-terminals-fixed_terminals