bigtreetech/PandaTouch #328: Reboots

Hondafave, 2025-07-23:
After the panda touch has gone into sleep mode, if you touch the screen to wake it up sometimes it reboots. I'm on 1.0 7 2 firmware.

Neuropsihopuppu, 2025-07-24:
Thats because the panda touch does not get enough power from the internal usb port of the printer

Hondafave, 2025-07-24:
I read that the panda touch needs 0.7 of an amp and the P1s puts out 1.5
amps so how come it's not enough? That's where the instructions tell you to
plug it in?

bl4ckm0onk3y, 2025-07-28:
> Thats because the panda touch does not get enough power from the internal usb port of the printer

I have it plugged into a cell phone charger with over 2A so it's not that, and it sometimes reboots the same way as OP described.

dolzhanskiyoa, 2025-07-28:
Same here/ FW  1.0.7.1, cell phone charger