bigtreetech/PandaTouch #331: Cannot connect to wifi networks with wpa2/wpa3 (Wifi 7)

voxpop9, 2025-07-29:
Please add the ability to connect the Panda Touch to Wifi with wpa2/wpa3 (wifi 7). Panda PwR is already able to do this.

voxpop9, 2026-08-21:
> In the latest released firmware, there have been optimizations for this section. Have you tried it, and has the issue been resolved?

Let me try it, as I had to get around this previously by switching to WPA1/2 Temporarily

voxpop9, 2026-08-21:
> > In the latest released firmware, there have been optimizations for this section. Have you tried it, and has the issue been resolved?
> 
> Let me try it, as I had to get around this previously by switching to WPA1/2 Temporarily

I am unable to disconenct the Panda Touch from the wifi to test if it can now connect with WPA2/3. How do I disconnect it without doing factory reset?

Lyle-68, 2026-08-21:
@voxpop9 You can try using your phone's hotspot as a temporary Wi-Fi network.
First, connect the Panda Touch to your phone's hotspot. Then switch back to your original WPA2/WPA3 Wi-Fi network and try connecting again.
This should allow you to test it without doing a factory reset.

voxpop9, 2026-08-27:
> [@voxpop9](https://github.com/voxpop9) You can try using your phone's hotspot as a temporary Wi-Fi network. First, connect the Panda Touch to your phone's hotspot. Then switch back to your original [...]

Sorry for the late reply. Can confirm it works now following your testing instructions. Thanks!

closed 2026-08-27 by voxpop9